chrisb2/pi_ina219 #2: Library is not compatible with Python 3

chrisb2, 2017-02-04:
These errors occur:
```
Traceback (most recent call last):
  File "example.py", line 4, in <module>
    from ina219 import INA219
  File "/home/pi/pi_ina219/ina219.py", line 7, in <module>
    import Adafruit_GPIO.I2C as I2C
ImportError: No module named 'Adafruit_GPIO'
```
and:
```
  File "example.py", line 14
    print "Bus Voltage    : %.3f V" % ina.voltage()
```

chrisb2, 2017-02-04:
To install dependencies for Python 3 on Raspberry Pi, this need to be executed:
`sudo python3 -m pip install git+git://github.com/chrisb2/pi_ina219.git`
this fixes the import error above.

closed 2017-02-04 by chrisb2

Losaccoj, 2017-09-09:
Thank you so much!! I was really stumped before I found this.

rhaejr, 2018-01-14:
I'm not sure if I am doing something wrong or what but this is not fixing the issue for me? Everything works great if i us Python 2.7, but everything goes to pot when I try it on Python 3.6.

chrisb2, 2018-01-15:
If you ran
`pip list`
then this gives you the packages install for Python 2, to see what is install for Python 3, you need:
`pip3 list`
Similarly to install the library for Python 3.6 you would need to use pip3. See  [INSTALLING PYTHON PACKAGES](https://www.raspberrypi.org/documentation/linux/software/python.md)

What error are you getting? 
Can you provide an example of the failing code?

regards,
Chris

david-harkey, 2018-02-24:
Similar to rhaejr above, this fix isn't working. I'm getting an error when I run that line:
```
Downloading/unpacking git+git://github.com/chrisb2/pi_ina219.git
  Cloning git://github.com/chrisb2/pi_ina219.git to /tmp/pip-vplibbw8-build
fatal: unable to connect to github.com:
github.com[0: 192.30.253.113]: errno=No route to host
github.com[1: 192.30.253.112]: errno=No route to host

  Complete output from command /usr/bin/git clone -q git://github.com/chrisb2/pi_ina219.git /tmp/pip-vplibbw8-build:

----------------------------------------
Cleaning up...
Command /usr/bin/git clone -q git://github.com/chrisb2/pi_ina219.git /tmp/pip-vplibbw8-build failed with error code 128 in None
Storing debug log for failure in /root/.pip/pip.log
```

I looked in the debug log that was mentioned, and here's the output:
```
------------------------------------------------------------
/usr/lib/python3/dist-packages/pip/__main__.py run on Sat Feb 24 22:28:30 2018
Downloading/unpacking git+git://github.com/chrisb2/pi_ina219.git
  Cloning git://github.com/chrisb2/pi_ina219.git to /tmp/pip-1zn5nw2s-build
  Found command 'git' at '/usr/bin/git'
  Running command /usr/bin/git clone -q git://github.com/chrisb2/pi_ina219.git /tmp/pip-1zn5nw2s-build
  Complete output from command /usr/bin/git clone -q git://github.com/chrisb2/pi_ina219.git /tmp/pip-1zn5nw2s-build:

----------------------------------------
Cleaning up...
Command /usr/bin/git clone -q git://github.com/chrisb2/pi_ina219.git /tmp/pip-1zn5nw2s-build failed with error code 128 in None
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1198, in prepare_files
    do_download,
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1361, in unpack_url
    unpack_vcs_link(link, loc, only_download)
  File "/usr/lib/python3/dist-packages/pip/download.py", line 344, in unpack_vcs_link
    vcs_backend.unpack(location)
  File "/usr/lib/python3/dist-packages/pip/vcs/__init__.py", line 240, in unpack
    self.obtain(location)
  File "/usr/lib/python3/dist-packages/pip/vcs/git.py", line 111, in obtain
    call_subprocess([self.cmd, 'clone', '-q', url, dest])
  File "/usr/lib/python3/dist-packages/pip/util.py", line 716, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command /usr/bin/git clone -q git://github.com/chrisb2/pi_ina219.git /tmp/pip-1zn5nw2s-build failed with error code 128 in None
```

Looks like I'm having issues cloning the repo. Any thoughts?

chrisb2, 2018-02-24:
It looks like from the error:
`fatal: unable to connect to github.com`
that github.com in not contactable by your machine, what happens if you do:
`ping github.com`
If this does not work, you may need to check your network setup.

Chris

david-harkey, 2018-02-25:
I think you're right, it seems to be a network issue. (I can connect to the internet via GUI apps or apt-get commands, but I can't ping or use git to clone)

In any case, I was able to install the package via `sudo pip3 install pi-ina219`. Thanks for your help!

GawiQ, 2018-03-01:
Hi 
I had this problem before and managed to fix it with
`sudo python3 -m pip install git+git://github.com/chrisb2/pi_ina219.git`
Adafruit_GPIO was indeed not listed on pip3 list.

Unfortunatelly I got some system upgrades and now the following error occurs again:
`  File "/home/pi/Desktop/python-lis3dh-master/LIS3DH_bus1.py", line 2, in <module>
    from Adafruit_GPIO import I2C
ImportError: No module named 'Adafruit_GPIO'`
but now when I checked the lists Adafruit_GPIO was present on both and installing dependecies for python 3 doesn't seem to work :/

Please help

chrisb2, 2018-03-02:
Try using pip3 to uninstall pi_ina219 and Adafruit_GPIO, check they are no longer listed by pip3 list, then re-install pi_ina219.

Chris

chrisb2, 2018-03-02:
> Thanks for response. I tried what you suggested but it didn't work. Still
> got the same error.

Please can you send me a simplied version of the  LIS3DH_bus1.py which
replicates this issue and I will give it a go myself. This will help to
isolate if the issue is in the program or your system.

regards,
Chris

chrisb2, 2018-03-03:
One more idea, did you try unistalling Adafruit_GPIO and then following
manual install instructions here:

https://github.com/adafruit/Adafruit_Python_GPIO

Chris

GawiQ, 2018-03-03:
[LIS3DH_bus1.py.gz](https://github.com/chrisb2/pi_ina219/files/1777999/LIS3DH_bus1.py.gz)
Manual installation didn't work. Do you think system reinstallation would work? Here is the file

chrisb2, 2018-03-04:
On a new installation of Raspbian I did:

   1. Enabled I2C in raspi-config
   2. Create a python test file containing just the two imports at the top
   of your file
   3. Ran my test file with python3 -> gave the same import error you get
   4. Installed Adafruit_GPIO with: pip3 install AdaFruit_GPIO
   5. Ran my test file again -> succeeded with no errors

So from this is does look like your system is to blame. So reinstalling
Rasbian will work.

Chris

GawiQ, 2018-03-04:
Did the same and now it works. 
Thanks man, really appreciate this. Awesome helping you did here :)