chrisb2/pi_ina219 #6: The library could not be imported in python >3.4

alcheagle, 2017-11-10:
I suggest adding 'Programming Language :: Python :: 3' inside ``setup.py`` classifiers instead of 'Programming Language :: Python :: 3.4'

chrisb2, 2017-11-11:
I had a look around and found this
https://packaging.python.org/tutorials/distributing-packages/#classifiers
which would seem to suggest it should perhaps be:

'Programming Language :: Python :: 2',
'Programming Language :: Python :: 2.7',
'Programming Language :: Python :: 3',
'Programming Language :: Python :: 3.4',

regards,
Chris

chrisb2, 2017-11-15:
I tested installing library on Raspbian Stretch:

`python -V`
gives 2.7.13
`sudo pip install pi-ina219`
succeeds.

`python3 -V`
gives 3.5.3
`sudo pip3 install pi-ina219`
succeeds.

Please can you provide more detail on the failure you observed.

thanks,
Chris

alcheagle, 2017-11-15:
The problem occurred while I imported them inside python code. I tested to import it inside of a docker container because I suspect that the problem is my python environment that is broken in some way. Now the import is working. I think I need to do a cleanup of my system. 

Sorry to bother and thanks,
Andrea.

closed 2018-04-30 by chrisb2