chrisb2/pi_ina219 #9: addressing i2c bus numbers > 1

eXodus1440, 2018-04-28:
Hi Chris,

I've been attempting to use your library with an Intel UpSquared and ran into an issue with the Intel's bus numbering system - their GPIO bus is 5 rather than the RPi default (assuming 1)

I made the following changes to ina219.py get it working on my project

`def __init__(self, shunt_ohms, max_expected_amps=None, busnum=None, address=__ADDRESS, log_level=logging.ERROR):`
`    self._i2c = I2C.get_i2c_device(address=address, busnum=busnum)`

Cheers,
Steve

chrisb2, 2018-04-29:
That looks like a good change, can you submit a Pull Request? This should not affect RPi since if None is passed as busnum to get_i2c_device() this method tries to determine default of the board, see [I2C](https://github.com/adafruit/Adafruit_Python_GPIO/blob/c543d1df9c0a71bafb9f0a1f9dceecd79a920e74/Adafruit_GPIO/I2C.py#L64). 

thanks,
Chris

eXodus1440, 2018-05-01:
Will do, will be later in the week tho

Cheers,
Steve

eXodus1440, 2018-05-16:
Hi Chris,

I'm trying to submit the PR but it's saying I don't have permissions, any ideas?

Just in case I was doing something silly, I also followed the below:
https://yangsu.github.io/pull-request-tutorial/

Regards,
Steve

chrisb2, 2018-05-17:
Do you mean you cannot create the PR?

Chris

chrisb2, 2018-05-17:
Assuming you forked my repo you would do "New pull request", add a comment
about your submission, then do " Create pull request"

chrisb2, 2018-06-26:
Version 1.2.0 containing this improvement has been upload to pypi: [pi-ina219](https://pypi.org/project/pi-ina219/)

closed 2018-06-26 by chrisb2