chrisb2/pi_ina219 #10: Guide until get data of voltage and current ina219 using raspberrypi

abeng94, 2018-05-16:
okay this LED in DIM condition (10% of brightness)
-voltage - below 1 V
Led in full brightness (100% of brightness)
-voltage - above 5V

I measure using multimeter.

can u guide me until I get this value using ina219.
I follow all video and this instruction . but get some error. 

that error say the python cant find that lib.

thank you

chrisb2, 2018-05-16:
It sounds like you executed your Python script and got an error like:
`Traceback (most recent call last):
  File "tmp.py", line 2, in <module>
    from ina219 import INA219
ImportError: No module named ina219
`
If so it seems like you have not installed the ina219 library using Python pip. 

If you are using Python 2 to execute your script (python command), then you should install with:
`sudo pip install pi-ina219`
you can check if its already installed with:
`pip list | grep ina219`

If you are using Python 3 to execute your script (python3 command), then you should install with:
`sudo pip3 install pi-ina219`
you can check if its already installed with:
`pip3 list | grep ina219`

regards,
Chris

chrisb2, 2018-06-26:
Closing for the moment, if you have further questions, please re-open.

closed 2018-06-26 by chrisb2

chrisb2, 2022-06-04:
> I have re-installed your library several times. No matter which way I call
> it, I get
> ModuleNotFoundError: No module named 'INA219'
>
> So frustrating!
> [image: 2022-06-02-222428_1920x1080_scrot]
> <https://user-images.githubusercontent.com/13710760/171733109-e8a8e15a-c8b2-4a78-968d-77aa6ac467ed.png>

Strange I agree. A couple of questions.

What Raspberry Pi is this?
What version of Raspbian is this?
If you execute the following what version do you see?

python3 -V

Chris