chrisb2/pi_ina219 #13: Johnny Five Plugin

msi-matthew, 2018-09-04:
What would it take to port this over to to work in the Johnny Five environment? (https://github.com/rwaldron/johnny-five) (http://johnny-five.io/)

chrisb2, 2018-09-05:
Porting the entire functionality would take you quite a bit of time, but
getting something basic going should be possible if you have learnt
Javascript doing other Johnney-Five programming.

Its clear from https://github.com/nebrius/raspi-io and
https://github.com/rwaldron/io-plugins that I2C on RPi is supported by J5.

To check you can talk to the ina219 you would just need to convert the
following methods to Javascript:
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L307
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L363

If that works I recommend moving on to try to read voltage as this does not
require calibration. The python methods required to be converted would then
be:
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L248
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L303
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L355
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L168
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L335
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L339
https://github.com/chrisb2/pi_ina219/blob/cf671f70974348ac202f613c6b090f05d4f2543b/ina219.py#L363

Another alternative might be to look at porting
https://github.com/brettmarl/node-ina219, but persueing  the same strategy
as above.

regards,
Chris

chrisb2, 2018-09-05:
Have you asked on J5 forums in any else has done this?

msi-matthew, 2018-09-05:
I did, and there seems to be some interest in a port.  After doing some more research, it seems like (https://www.npmjs.com/package/ina219) might be a good place to start.

closed 2018-10-15 by chrisb2