chrisb2/pi_ina219 #14: Calculation of "current" for negative "shunt_voltage"

davidbroster, 2018-10-14:
when the shunt voltage is positive all is as expected (voltage range=12-14V, shunt 16mV, current 160mA)
when the shunt voltage is negative (or the Vin- and Vin+) are swapped (same voltage range, shunt -16mV) then the current is given as about -1100mA.  Is this expected behaviour for the library (v1.2.0).  I'm puzzled as I thought the INA219 could handle positive and negative shunt voltages.  The application is measuring charge and discharge of a 12 volt battery at minute intervals with the INA219 in series with the battery positive terminal.  All aspects of the INA219 are standard except I set the voltage range to 16 Volts and the Max current to 2,0 Amps.

chrisb2, 2018-10-15:
This should work, l will have to reconnect my pi to ina219 to test this, it
may take me a few days.

Chris

davidbroster, 2018-10-15:
Thanks Chris

There is no panic but it would be good to know. I agree it should work, in parallel I’ll look at the GIT code.  At this stage all I can verify is that I measured Vin+, Vin-  and Vshunt (default 0.1 Ohm) with my autoranging multimeter and the values as read from the 219 call are always correct. It is just that the calculation of current (and power of course) are nonsense when Vshunt is -ve.  If the sun would shine (😊) I could see what happens then the battery is charged from the solar panel and should be similar to me swapping the wires over. It is a new installation on a cloudy day which is why I swapped the wires and saw the issue in the first place.

More later …..

David

________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom
GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster

davidbroster, 2018-10-16:
Hi (again) Chris,

The sun is shining a little today so I checked the readings ….. peculiar indeed ….  I’ve included some screenshots ….
First the script (standard) so no jiggery-pockery going on behind the scenes …

[cid:image001.png@01D4654A.3E98E5E0]

And then some readings ….. taken a couple of minutes apart

[cid:image002.png@01D4654A.3E98E5E0]

The first one appears correct 😊
The second one demonstrates the weirdness ☹
The third one is consistently weird …..

When it goes dark the shunt_voltage will be positive and I’ll check again for weirdness, but I never saw any last night or yesterday -  until the sun came out and the polarity changed from discharge to charge !

Am I “nuts” ?

David
________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom
GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster

davidbroster, 2018-10-16:
Chris,

Note I’d added a “reset” into the script (commented out), after sending the last email I uncommented and ran 10 times ….. and the readings for current are still “silly” …..

David

________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom

GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster

[cid:image001.png@01D4654B.4429A700]

[cid:image002.png@01D4654B.4429A700]

chrisb2, 2018-10-18:
For some reason the images did not come through. The time I remember
getting strange readings was when I did not have a solid common ground
between the GND of the ina219 and the battery -ve. If you are unclear how
to achieve this just search for "ina219 common ground".

Still planning on setting up to do some testing myself.

Chris

davidbroster, 2018-10-18:
The images were .png  files from screenshots saved to my Dropbox …. So I’ve turned them into a pdf (attached)

There is a common ground and it is pretty solid, my meter suggests just a few microvolts between the various ground terminals …..

The other thing I’ve noticed is that when the current looks sort of “normal” it usually but not always V/R which should be 10*V(mV) in mA for 0.1 Ohm shunt.
I don’t understand why it isn’t always shunt_millivolts*10 mA  ?

Thanks in anticipation 😊   David

________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom

GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster

<mailto:notifications@github.com%3cmailto:%0b>> notifications@github.com<mailto:notifications@github.com>>>
<mailto:pi_ina219@noreply.github.com%3cmailto:%0b>> pi_ina219@noreply.github.com<mailto:pi_ina219@noreply.github.com>>>
<mailto:notifications@github.com%0b>> <mailto:notifications@github.com>>

chrisb2, 2018-10-19:
Here are some figures from my system in a google spreadsheet;

https://docs.google.com/spreadsheets/d/1ncwnngbgqmy371HEMcOB8UHByTu9iFI-wHZcdQNkxiM/edit?usp=sharing

The load was a 12V/5W car bulb powered by a bench power supply. No luck replicating your issue so far. Can you send a circuit diagram?

Still did not get the PDF, it does not seem to be attached to the thread in Github, unless they are hidden in strange place.

Your question about V/R calculation not seeming right does not seem to be present in my results, taking the first row the shunt voltage is 31.78mV (voltage dropped across 0.1 Ohms) and the bus current is 317mA.

Chris

davidbroster, 2018-10-19:
Hi Chris, very convincing – I’ll explore more.

My ina219 is down the garden in the greenhouse as part of a solar watering system which uses as PiZeroW and an upside-down satellite positioner so the solar panel tracks the sun to charge the battery.  Lots of fun putting it together and figuring out the positioner DISQC codes as 22Khz tone-bursts  - but then in anticipation of sunless days I wanted to measure the cumulated battery charge / discharge current based on 1 minute snapshots 24/7.

I couldn’t (and still can’t) understand my readings, so I’ll hook up another ina219 in my lab and test as you did.  In fact I’ll replace the remote ina219 with another and  bring that one in and use that in my bench tests. I feel a right dumbo as a retired electronics engineer this should be easy peasy, but I’ll get to the bottom of it and get back to you.  We are away from home (hence the greenhouse system) so it’ll most likely be during November.

Thanks so far for helping me chase the gremlins-in-the-greenhouse 😊

David
p.s. I simply attached my pdf to my reply which I sent from Outlook on my PC  ….. I’ll look to go to github and upload it …..

________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom
GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster

https://docs.google.com/spreadsheets/d/1ncwnngbgqmy371HEMcOB8UHByTu9iFI-wHZcdQNkxiM/edit?usp=sharing

davidbroster, 2018-11-09:
Hi Chris, sorry for the silence (due to other stuff happening around me).  I set-up a separate 2nd PiZw and found the measurements to be entirely consistent and fine similar to your results.  So I’ve now split the current measurements away from the main daemon that controls the system down my garden and setup a separate daemon in an effort to isolate the issue., and I’ve moved the 2nd PiZw down the garden to sit alongside the original system (all connections in parallel except for the shunts which are in series.  Both systems  append measurements to their own a csv files noting all the read parameters every 20 seconds.  Typical system voltage is 12-14 volts with typical current around 200mA (1500mA when the watering pump is on).  It’s been running for a few hours now so I took a peek and just looked at a single column that is the calculated current (ma) divided by the shunt_voltage (mV) with an expected answer ~10.  The 2nd Pi shows 9.9 – 10.1 even when the pump is on.  The original Pi shows the same values more or less until the pump comes on when the shunt voltage rises from 18mV to 150mV and then the calculated ratio changes to about 3 ! ! ! ! This is weird since the current through the shunt is a simple calculation based on the shunt voltage …..
Early days but I want to find out the quirk wherever, whatever ….. which lies within the original Pi setup.  Interestingly each measured differential shunt voltages correspond between the two systems, it is just the current resulting from i = ina.current() on the original system goes daft.

More on this after I’ve looked at the two sets of measurements in a 24 hour .csv file (hopefully with some sun and hence battery charging tomorrow 😊)

David
________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom
GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster
________________________________

From: David Broster
Sent: 19 October 2018 11:57
To: 'chrisb2/pi_ina219' <reply@reply.github.com>
Subject: RE: [chrisb2/pi_ina219] Calculation of "current" for negative "shunt_voltage" (#14)

Hi Chris, very convincing – I’ll explore more.

My ina219 is down the garden in the greenhouse as part of a solar watering system which uses as PiZeroW and an upside-down satellite positioner so the solar panel tracks the sun to charge the battery.  Lots of fun putting it together and figuring out the positioner DISQC codes as 22Khz tone-bursts  - but then in anticipation of sunless days I wanted to measure the cumulated battery charge / discharge current based on 1 minute snapshots 24/7.

I couldn’t (and still can’t) understand my readings, so I’ll hook up another ina219 in my lab and test as you did.  In fact I’ll replace the remote ina219 with another and  bring that one in and use that in my bench tests. I feel a right dumbo as a retired electronics engineer this should be easy peasy, but I’ll get to the bottom of it and get back to you.  We are away from home (hence the greenhouse system) so it’ll most likely be during November.

Thanks so far for helping me chase the gremlins-in-the-greenhouse 😊

David
p.s. I simply attached my pdf to my reply which I sent from Outlook on my PC  ….. I’ll look to go to github and upload it …..

________________________________
David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom

GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    http://maps.google.com/maps?q=@52.270758,0.762636

House:               +44 (0)1745 776 126
Mobile David        +44 (0)7557 359 095     david@broster.net<mailto:david@broster.net>              Skype: davidbroster
Mobile Margaret    +44 (0)7557 359 417     margaret@broster.net<mailto:margaret@broster.net>       Skype: mbroster

https://docs.google.com/spreadsheets/d/1ncwnngbgqmy371HEMcOB8UHByTu9iFI-wHZcdQNkxiM/edit?usp=sharing

chrisb2, 2019-02-09:
Closing, but re-open if you need further assistance.

Chris

closed 2019-02-09 by chrisb2