chrisb2/pi_ina219 #15: Current overflow false positive

Gbassous, 2019-02-04:
I'm testing the INA219 with a raspberry pi 3B with a voltage source and cannot read above 0,32 A, despite it being possible to get up to 3,2 A, or 0,32 V on the shunt. I consistently get current overflow errors when reaching 0,32 A / 0,032 V on the shunt (tested with a high precision multimeter). Tested with both auto-gain and manual, and both voltage ranges. I find it odd that the amp value raising the overflow is the voltage value that would produce max current on the shunt.

Gbassous, 2019-02-05:
To get it working for now, i set __OVF = 0, since I am well aware of the tension and current values I am working with. I believe that somewhere in the current overflow detection procedure the comparison is being made against max tension instead of max current.

Gbassous, 2019-02-05:
This is what i get with logging and both auto-gain and manual gain:
2019-02-05 15:15:46,955 - INFO - INA219 gain set to 0.32V
2019-02-05 15:15:46,956 - INFO - INA219 calibrate called with: bus max volts: 32V, max shunt volts: 0.32V, max expected amps: 1.600A
2019-02-05 15:15:46,956 - INFO - INA219 max possible current: 3.200A
2019-02-05 15:15:46,957 - INFO - INA219 max expected current: 1.600A
2019-02-05 15:15:46,957 - INFO - INA219 current LSB: 4.878e-05 A/bit
2019-02-05 15:15:46,957 - INFO - INA219 power LSB: 9.756e-04 W/bit
2019-02-05 15:15:46,957 - INFO - INA219 max current before overflow: 1.5984A
2019-02-05 15:15:46,958 - INFO - INA219 max shunt voltage before overflow: 159.8390mV
2019-02-05 15:15:46,958 - INFO - INA219 calibration: 0x20cc (8396)
press s for sample, e to save and end, k to end without saving: s
2019-02-05 15:15:47,893 - INFO - INA219 Current overflow detected - attempting to increase gain
2019-02-05 15:15:47,896 - INFO - INA219 gain is currently: 0.32V
2019-02-05 15:15:47,897 - INFO - INA219 Device limit reach, gain cannot be increased
Traceback (most recent call last):
  File "power_calib.py", line 18, in <module>
    sample = numpy.array([ina.voltage(),ina.shunt_voltage()])
  File "/home/pi/.virtualenvs/py3cv4gpio_mestrado/lib/python3.5/site-packages/ina219.py", line 194, in shunt_voltage
    self._handle_current_overflow()
  File "/home/pi/.virtualenvs/py3cv4gpio_mestrado/lib/python3.5/site-packages/ina219.py", line 221, in _handle_current_overflow
    self._increase_gain()
  File "/home/pi/.virtualenvs/py3cv4gpio_mestrado/lib/python3.5/site-packages/ina219.py", line 246, in _increase_gain
    raise DeviceRangeError(self.__GAIN_VOLTS[gain], True)
ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached

I'm running 2 V / 0.36 A through my circuit (according to the power supply indicators), with a 5 Ohm load. With a voltmeter I get 0.038 V in the shunt. If I run a current lower than 0.32 A it works without a problem. By setting __OVS = 0, from what I understood from your code, it doesn't flag any overflows and I am able to measure currents over 0.32 A with manual gain. What I find oddly coincidental (which means it probably isn't) is that it's detecting overflow when the current is at the value for max voltage in the shunt.

Regards,
Guilherme

chrisb2, 2019-02-05:
This does seem puzzling. I did some testing on my system with the following test program, which I think has the same configuration parameters that you using (at least log output is identical):
```
#!/usr/bin/env python
from ina219 import INA219
from ina219 import DeviceRangeError
import logging

SHUNT_OHMS = 0.1

def read():
    ina = INA219(SHUNT_OHMS, 1.6, log_level=logging.INFO)
    ina.configure(ina.RANGE_32V, ina.GAIN_8_320MV)

    print("Bus Voltage: %.3f V" % ina.voltage())
    try:
        print("Bus Current: %.3f mA" % ina.current())
        print("Power: %.3f mW" % ina.power())
        print("Shunt voltage: %.3f mV" % ina.shunt_voltage())
    except DeviceRangeError as e:
        # Current out of device range with specified shunt resister
        print(e)

if __name__ == "__main__":
    read()
```
My test load is a little different (its a 12V/21W car bulb) which results in my being able to generate the same current as you, but at a higher voltage. The higher voltage should not make a difference.

With the current just under the limit indicated in the logging (1.594A), I got:
```
2019-02-06 12:00:57,719 - INFO - INA219 gain set to 0.32V
2019-02-06 12:00:57,723 - INFO - INA219 calibrate called with: bus max volts: 32V, max shunt volts: 0.32V, max expected amps: 1.600A
2019-02-06 12:00:57,726 - INFO - INA219 max possible current: 3.200A
2019-02-06 12:00:57,730 - INFO - INA219 max expected current: 1.600A
2019-02-06 12:00:57,739 - INFO - INA219 current LSB: 4.878e-05 A/bit
2019-02-06 12:00:57,743 - INFO - INA219 power LSB: 9.756e-04 W/bit
2019-02-06 12:00:57,759 - INFO - INA219 max current before overflow: 1.5984A
2019-02-06 12:00:57,762 - INFO - INA219 max shunt voltage before overflow: 159.8390mV
2019-02-06 12:00:57,765 - INFO - INA219 calibration: 0x20cc (8396)
Bus Voltage: 10.884 V
Bus Current: 1584.537 mA
Power: 17252.683 mW
Shunt voltage: 158.490 mV
```
and with the current just over the limit:
```
2019-02-06 12:01:27,908 - INFO - INA219 gain set to 0.32V
2019-02-06 12:01:27,912 - INFO - INA219 calibrate called with: bus max volts: 32V, max shunt volts: 0.32V, max expected amps: 1.600A
2019-02-06 12:01:27,916 - INFO - INA219 max possible current: 3.200A
2019-02-06 12:01:27,919 - INFO - INA219 max expected current: 1.600A
2019-02-06 12:01:27,929 - INFO - INA219 current LSB: 4.878e-05 A/bit
2019-02-06 12:01:27,932 - INFO - INA219 power LSB: 9.756e-04 W/bit
2019-02-06 12:01:27,941 - INFO - INA219 max current before overflow: 1.5984A
2019-02-06 12:01:27,949 - INFO - INA219 max shunt voltage before overflow: 159.8390mV
2019-02-06 12:01:27,952 - INFO - INA219 calibration: 0x20cc (8396)
Bus Voltage: 11.540 V
Current out of range (overflow), for gain 0.32V
```
So my system is not replicating the issue you have.  Can you draw and attach the circuit diagram for your  setup?

regards,
Chris

chrisb2, 2019-02-06:
I attached my test circuit:
[image]

chrisb2, 2019-04-19:
Closing this issue, please re-open if you require further assistance.

closed 2019-04-19 by chrisb2

wojciechczyz, 2022-02-25:
I have got exactly the same problem when trying to install Raspberry PI i52 UPS Plus HAT. Device is called EP-0136
https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136

 I am getting following error:

------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.956 V
Raspberry Pi Current Current Consumption: 1180.656 mA
Raspberry Pi Current Power Consumption: 5843.566 mW
------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/bin/upsPlus.py", line 40, in <module>
    batt_current = ina_batt.current()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 196, in current
    self._handle_current_overflow()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 246, in _handle_current_overflow
    self._increase_gain()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 271, in _increase_gain
    raise DeviceRangeError(self.__GAIN_VOLTS[gain], True)
ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached
UPS Plus Installation is Incomplete! ... failed!
Please visit wiki for more information:.
-----------------------------------------------------.
https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.
-----------------------------------------------------.