chrisb2/pi_ina219 #17: cannot import name INA219

syjsxxjy, 2019-04-26:
```
pi@raspberrypi:~/shabi/python$ python ina219.py
Traceback (most recent call last):
  File "ina219.py", line 4, in <module>
    from ina219 import INA219
  File "/home/pi/shabi/python/ina219.py", line 4, in <module>
    from ina219 import INA219
ImportError: cannot import name INA219
```
and I tried to pip3 install it but same error occurred. How can I fix it?

chrisb2, 2019-05-01:
Sorry for delay in replying, I was ill last week. 

So initially ran the following and it succeeded?
`sudo pip install pi-ina219`

You can verify if this worked with:
`pip list | grep ina219`

It should return:
`pi-ina219 (1.2.0)`

I presume 'ina219.py' is a test program can you post the contents?

Python 2 or 3 should work, you just have to use the matching pip command 'pip' for Python 2 and 'pip3' for Python 3.

regards,
Chris

syjsxxjy, 2019-05-11:
[image]
[image]
[image]
Yes, I can install it by pip and it returns pi-ina219,but cannot import name....

syjsxxjy, 2019-05-13:
> Sorry for delay in replying, I was ill last week.
> 
> So you initially ran the following and it succeeded?
> `sudo pip install pi-ina219`
> 
> You can verify if this worked with:
> `pip list | grep ina219`
> 
> It should return:
> `pi-ina219 (1.2.0)`
> 
> I presume 'ina219.py' is a test program can you post the contents?
> 
> Python 2 or 3 should work, you just have to use the matching pip command 'pip' for Python 2 and 'pip3' for Python 3.
> 
> regards,
> Chris

I just give up . I copied your code file to my project and import works properly.

chrisb2, 2019-05-14:
OK, good to hear you got it working.

closed 2019-05-14 by chrisb2

remoengineer, 2019-09-15:
The interpreter picked up your ina219.py as the module instead of your test program.  Simply rename the test program to something like ina219_test.py and it should get well.