chrisb2/pi_ina219 #19: Use named logger instead of root

JennaSys, 2019-11-02:
Would it be possible to update ina219.py to use a named logger rather than the root logger?   Being that this library is usually imported, `logger.basicConfig()` has likely been called by the main module already, so the `basicConfig()` call in `__init__()` has no effect (IOW the log level will not be changed at that time if it is different from the main module)

Use something like:
`logger = logging.getLogger(__name__)`
then change log calls to use `logger.info()` instead of `logging.info()` for example.

I can fix it and do a PR if you prefer.  Thanks.

chrisb2, 2019-11-02:
Great idea please submit pr.

Chris

JennaSys, 2019-11-03:
PR submitted.

chrisb2, 2019-11-09:
PR merged and new 1.3.0 release created and uploaded to PyPi, see https://pypi.org/project/pi-ina219/1.3.0/

thanks,
Chris

closed 2019-11-16 by chrisb2

rrnicolay, 2023-03-02:
I can't stop the module from printing debug messages. What can I do?
My code:

`import logging
from ina219 import INA219, DeviceRangeError
logging.getLogger('ina219').setLevel(logging.ERROR)

class CurrentReader():
  # Default value of shunt resistor on INA219
  SHUNT_OHMS = 0.1
  # Max expected amps read by INA219 sensor
  MAX_EXPECTED_AMPS = 3.0

  def __init__(self):
    self.ina = INA219(self.SHUNT_OHMS, self.MAX_EXPECTED_AMPS, busnum=1, log_level=logging.ERROR)`

chrisb2, 2023-03-06:
It's not good practice to comment on a closed issue, please open a new one.

Thanks
Chris