chrisb2/pi_ina219 #21: getting  [Errno 121] Remote I/O error?

SmartTransp, 2019-11-05:
Hello m 
I don't know if this is the place to ask - but I couln't find answer anywhere.

I have done everything is said in the guide : (except the LCD part)
https://www.rototron.info/raspberry-pi-ina219-tutorial/ 

download and try to run the code (without the LCD) 
but I'm gettting the
`[Errno 121] Remote I/O error `
what could be the reason for it 

when I run the scanner - I can see the INA219 

`sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
`

this is the code I'm using 
`from ina219 import INA219, DeviceRangeError
from time import sleep

SHUNT_OHMS = 0.1
MAX_EXPECTED_AMPS = 0.6
ina = INA219(SHUNT_OHMS, MAX_EXPECTED_AMPS)
ina.configure()

def read_ina219():
    try:
        print ('Bus Voltage: {0:0.2f}V'.format(ina.voltage()))
        print ('Bus Current: {0:0.2f}mA'.format(ina.current()))
        print ('Power: {0:0.2f}mW'.format(ina.power()))
        print ('Shunt Voltage: {0:0.2f}mV\n'.format(ina.shunt_voltage()))
    except DeviceRangeError as e:
        # Current out of device range with specified shunt resister
        print(e)

while 1:
    read_ina219()
    sleep(1)`

and this is the all error I'm getting

`python Documents/TestINA219_a.py 
Traceback (most recent call last):
  File "Documents/TestINA219_a.py", line 7, in <module>
    ina.configure()
  File "/usr/local/lib/python2.7/dist-packages/ina219.py", line 165, in configure
    self._max_expected_amps)
  File "/usr/local/lib/python2.7/dist-packages/ina219.py", line 283, in _calibrate
    self._calibration_register(calibration)
  File "/usr/local/lib/python2.7/dist-packages/ina219.py", line 329, in _calibration_register
    self.__write_register(self.__REG_CALIBRATION, register_value)
  File "/usr/local/lib/python2.7/dist-packages/ina219.py", line 361, in __write_register
    self._i2c.writeList(register, register_bytes)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/I2C.py", line 127, in writeList
    self._bus.write_i2c_block_data(self._address, register, data)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_PureIO/smbus.py", line 294, in write_i2c_block_data
    self._device.write(data)
IOError: [Errno 121] Remote I/O error`

is there any way you can assit or find the problem ?
Thanks ,

chrisb2, 2019-11-08:
I have no specific fix, a bit of fishing on the internet revealed that this error is due to smbus not being able to talk to the device. Some connection is present since you can see the device on the i2c bus. All I can suggest is:

- re-checking your wiring
- are your wires long? If so try with shorter wires.
- try a slower bus speed - [see](https://www.abelectronics.co.uk/kb/article/1/i2c--smbus-and-raspbian-linux)
- if you have another Pi, or an Arduino, or ESP8266/ESP32 try the INA219 with them.

By the way what version of Raspberry Pi are you using?

regards,
Chris

SmartTransp, 2019-11-17:
1. the wires are OK (double check 10 times).
2. the wire length is about 10 CM.

I saw the problem is when I connect the GND that everything stop working . 
when I don't connect the GND to the PI - he show me the scan but unabel to read 
when I connect the GND - the sacn take about 3 min , and he doesn't show me the device.

I saw that when I connect it to an Arduino:
when I connect the GND I get this message from the computer : usb over current status detected(which mean there is a problem)

when I disconnect the GND - no problem running the Arduino with the INA219 , but it doesn't read the device I get 32V all the time.

this is why I thought maybe I need to add some pullups or something ?
this is what Arduino forum suggest also  - do you ever saw this problem ?

I'm using PI4 , try it also with PI 3 and PI 2  - so I don't think it's a PI issue 

just to give you more information : I have try 5 INA219 - with all of them I have the same problem.

any idea or thought?

Thanks ,

JennaSys, 2019-11-17:
A ground wire MUST be connected between the PI and the INA219 to establish a voltage reference.  This is the only way the voltage levels for the signals can be reliably sent.

Some INA219 devices have been known to have manufacturing defects (I had a few myself).  Without it being connected to anything, and no power applied, check the resistance between the Vcc and Gnd terminals of the INA219.  On the defective ones I had there was a near short (< 10 ohms).

chrisb2, 2019-11-19:
Assuming you have followed exactly the circuit described in the tutorial, I would very carefully check all your wires, perhaps you have a dodgy one? Perhaps just replace them all? I have two devices and succeeded in getting them working pretty much straight away.

chrisb2, 2020-03-29:
Closing, re-open if required.

closed 2020-03-29 by chrisb2

DotEXEGaming, 2022-11-22:
Re-opening this as I have the same issue. Wires aren't too long (~10cm), INA219 board is hooked up correctly. Have troubleshooted using what I've found here and other sources. Please get back to me as soon as possible.

DotEXEGaming, 2022-11-22:
```
#!/usr/bin/env python
from ina219 import INA219
from ina219 import DeviceRangeError
import time

SHUNT_OHMS = 0.1
MAX_EXPECTED_AMPS = 0.2

def read():
    ina = INA219(SHUNT_OHMS, MAX_EXPECTED_AMPS, busnum=1)
    ina.configure(ina.RANGE_16V)

    print("Bus Voltage: %.3f V" % ina.voltage())
    try:
        print("Bus Current: %.3f mA" % ina.current())
        print("Power: %.3f mW" % ina.power())
        print("Shunt voltage: %.3f mV" % ina.shunt_voltage())
    except DeviceRangeError as e:
        # Current out of device range with specified shunt resistor
        print(e)

if __name__ == "__main__":
    read()
```
attached code*