chrisb2/pi_ina219 #22: Unable to get ADC_16SAMP (or any of the others to work)

skibum-za, 2019-12-10:
I didn't have any issues with the first two sample programs. So, I thought I'd try changing the sampling options. So I changed the following :- 

ina.configure(ina.RANGE_16V, ina.ADC_16SAMP)

This seemed correct (although I'm sure you'll tell me I'm doing it wrong). Howeever, python gives me an error when doing this :- 

`$ python3 ina219_2.py`

`Traceback (most recent call last):`
`  File "ina219_2.py", line 29, in <module>`
`    read()`
`  File "ina219_2.py", line 13, in read`
`    ina.configure(ina.RANGE_16V, ina.ADC_16SAMP)`
`  File "/usr/local/lib/python3.7/dist-packages/ina219.py", line 164, in configure`
`    self.logger.info('gain set to %.2fV' % self.__GAIN_VOLTS[self._gain])`
`IndexError: list index out of range`

Oddly, if I also add :- 

`ina.configure(ina.RANGE_16V, ina.ADC_12BIT, ina.ADC_16SAMP)`

It then 'works'. I put works in quotes, as it then starts giving me overflow errors :- 

`Bus Voltage: 8.440 V`
`Current out of range (overflow), for gain 0.32V`

If I take out both out and just leave the line as :- 

`ina.configure(ina.RANGE_16V)`

then it works fine :- 

`pi@pizero-i2c:~ $ python3 ina219_2.py`

`Bus Voltage: 8.360 V`
`Bus Current: 1204.098 mA`
`Power: 10294.634 mW`
`Shunt voltage: 121.250 mV`

`Bus Voltage: 8.520 V`
`Bus Current: 1305.171 mA`
`Power: 12634.146 mW`
`Shunt voltage: 166.960 mV`

`Bus Voltage: 8.356 V`
`Bus Current: 1352.000 mA`
`Power: 11112.195 mW`
`Shunt voltage: 119.970 mV`

`Bus Voltage: 8.476 V`
`Bus Current: 1294.829 mA`
`Power: 11791.220 mW`
`Shunt voltage: 142.850 mV`

chrisb2, 2019-12-12:
The signature of the [configure() method](https://github.com/chrisb2/pi_ina219/blob/master/ina219.py#L123) means that the arguments (with defaults) are:

`(voltage_range=RANGE_32V, gain=GAIN_AUTO, bus_adc=ADC_12BIT, shunt_adc=ADC_12BIT)`

So:
`ina.configure(ina.RANGE_16V, ina.ADC_16SAMP)`

Is passing the sampling rate to the gain argument and so is invalid.

`ina.configure(ina.RANGE_16V, ina.ADC_12BIT, ina.ADC_16SAMP)`

Is again passing a sampling rate to the gain argument, so is likewise invalid. It sort of works because the values of these arguments are integers.

I think what you probably have in mind is:

`ina.configure(ina.RANGE_16V, ina.GAIN_AUTO, ina.ADC_16SAMP, ina.ADC_16SAMP)`

regards,
Chris

skibum-za, 2019-12-12:
Many thanks Chris. I was pretty sure it was 'user error'.

closed 2019-12-12 by skibum-za