chrisb2/pi_ina219 #28: Convert from Adafruit_GPIO library (which is archived) to smbus2 library

chrisb2, 2021-11-17:
The [Adafruit_GPIO](https://github.com/adafruit/Adafruit_Python_GPIO) library is now archived and so does not correctly support later versions of Raspberry Pi and OS. The plan is to replace this with [smbus2](https://github.com/kplindegaard/smbus2).

This stacktrace occurred on a Raspberry Pi 4 with 'Bullseye' Raspberry Pi OS:
```
File "/usr/local/lib/python3.9/dist-packages/ina219.py", line 116, in __init__
    self._i2c = I2C.get_i2c_device(address=address, busnum=busnum)
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_GPIO/I2C.py", line 63, in get_i2c_device
    busnum = get_default_bus()
  File "/usr/local/lib/python3.9/dist-packages/Adafruit_GPIO/I2C.py", line 55, in get_default_bus
    raise RuntimeError('Could not determine default I2C bus for platform.')
RuntimeError: Could not determine default I2C bus for platform.
```
A work-around for this is described in the [readme](https://github.com/chrisb2/pi_ina219#i2c-bus-number).

eliran-n, 2022-02-13:
Hi Chris, 

I would like to use the library of the ina219 on python and I have some questions please.,
Basically, I'm running Pyfirmata (StandardFirmata) on my Arduino and I'm controlling some motor via Python.
I want to connect the INA219 to my Arduino and read the current samples from Python to display it on my App.
Do I have to use Raspberrypi board ? Because I am more familiar with Arduino. 
It is possible to use this library to do so? 

Thanks in advance, eliran

chrisb2, 2022-02-14:
@eliran-n - please raise this as a new issue, a PR is not a suitable place to ask unrelated questions. Thx

eliran-n, 2022-02-14:
> @eliran-n - please raise this as a new issue, a PR is not a suitable place to ask unrelated questions. Thx

Sorry, I was connected from a mobile device and I didn't find a way to contact you.
I post my question as a new issue. Thank you