chrisb2/pi_ina219 #34: OSError: [Errno 121] Remote I/O error

DotEXEGaming, 2022-11-23:
As indicated by the title I am getting this error.
All code and error message is here: https://pastebin.com/KBh53Uq3
Running on a Raspberry Pi 4B 4GB.

Any help would be greatly appreciated, thanks!

DotEXEGaming, 2022-11-23:
[image]
circuit diagram *

DotEXEGaming, 2022-11-24:
@chrisb2 sorry to at you but this is important for its use

chrisb2, 2022-11-25:
Code and circuit look good. Please can you check that the device is visible on the i2c bus with:
`sudo i2cdetect -y 1`
as explain in [https://learn.adafruit.com/scanning-i2c-addresses/raspberry-pi](https://learn.adafruit.com/scanning-i2c-addresses/raspberry-pi), you should see it at address 0x40.

Chris

DotEXEGaming, 2022-11-25:
I2C is enabled and when running `i2cdetect -y 1` the sensor is detected on 0x40

chrisb2, 2022-11-25:
Searching in Google it seems the `OSError: [Errno 121] Remote I/O error` error is a pain to track down. Here are a few questions, sorry if you have already thought of some of these:

1. How long are leads between the Pi and the ina219?
2. Is the circuit diagram accurate or just a representation (sometimes the issue is something small)?

**Things to Try**
Add the following delay before you call `ina.configure()' in your program:
`utime.sleep_ms(10)'

Verify all the wires to the device are good and joints are solid? Are they soldered (dry joints?), or is this a prototype board (dodgy connection?)?

DotEXEGaming, 2022-11-26:
The leads are bout 10cm long, and the circuit diagram is accurate minus the light and 12V battery. 
when running `cat /etc/os-release`:
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I'm having issues with installing utime so i used `time.sleep(0.01)` instead but no luck.
All wires are solid dry joints.

chrisb2, 2022-11-27:
Using a Pi 4B and latest Bullseye and a simple circuit like yours except bulb is a LED and series resistor powered from Pi 5V, I can make a measurement using _example.py_ with _busnum=1_. I think it most likely the issue is something to do with your circuit, can you try the following to try and eliminate where:
1. Replace your existing load under measurement with an LED and resistor power from Pi 5V (so same as my setup). Does this work?
2. If it does not work, that might indicate an issue with the ina219 breakout itself. If you have spare, trying swapping to it in the LED circuit.
3. If it does work, then there must be some issue with your actual load, perhaps you can use a process of elimination to try and work out the problem? If you have a logic analyzer this might be the time to use it.

Chris

DotEXEGaming, 2022-11-27:
1. Same error.
2. The LED lights but the error persists.
3. I have not got any spare (soldered) boards currently.
Also the LED only lit when connecting to the screw terminals not the header (they should be the same?)

chrisb2, 2022-11-28:
Header should work same as terminals. If you use a magnifying glass do the tracks look OK, or use a multimeter to check tracks.

Given LED circuit is not working its beginning to sound like your ina219 has a problem. So we are clear, below is a sketch of my LED circuit.

[image]

Chris

DotEXEGaming, 2022-11-28:
To be clear, the LED does work now, yes, and I am using a separate power source with the ground shared with the pi.

chrisb2, 2022-11-28:
So you can measure current and voltage with LED circuit?

DotEXEGaming, 2022-11-28:
The LED lights, but the board gives the same error.

chrisb2, 2022-11-28:
So I think your ina219 has a problem, I have never seen this simple led circuit fail with my library. The LED is always going to light unless the shunt resistor is dead, the ina219 chip on the breakout is another matter. I think you need to obtain a new breakout.

DotEXEGaming, 2022-11-28:
Oh alright. I'll resolder another one and try then. Probably later today.

DotEXEGaming, 2022-11-28:
After soldering a new board and closely inspecting the old one to see the ground pin wasn't even soldered on, it works! Thank you so much. Have a great day/evening what be it.

closed 2022-11-28 by DotEXEGaming