chrisb2/pi_ina219 #38: No change in resolution or time when changing ADC samples and bit

csreades, 2023-05-06:
Testing with my own code and also the performance test script on Raspberry Pi 4. Using a 200 A 75 mV shunt (0.000375 ohms), no change in collection speed or acuracy of reading when changing ADC sample depth from 0 - 15.

chrisb2, 2023-05-06:
Interesting! If you look at P20 of the [datasheet](https://www.ti.com/lit/ds/symlink/ina219.pdf) the timings are described for all the settings. I would have thought the slowest setting at 68ms would have been noticeably different to the fastest. I am going to have play.

Perhaps you can attach your program, so we end up testing in a comparable way.

regards,
Chris

chrisb2, 2023-05-06:
I can replicate your result with [performance-test.py](https://github.com/chrisb2/pi_ina219/blob/master/performance-test.py). I had to add `busnum=1` to make it run, so:
```
ina = INA219(SHUNT_OHMS, MAX_EXPECTED_AMPS, log_level=logging.INFO, busnum=1)
```
then with either:
```
ina.configure(ina.RANGE_16V, ina.GAIN_AUTO, ina.ADC_128SAMP, ina.ADC_128SAMP)
```
or
```
ina.configure(ina.RANGE_16V, ina.GAIN_AUTO, ina.ADC_9BIT, ina.ADC_9BIT)
```
I get ~580uS.

Using `example.py` and setting debug logging with config:
```
ina.configure(ina.RANGE_16V, ina.GAIN_AUTO, ina.ADC_128SAMP, ina.ADC_128SAMP)
```
I can see the config register is set correctly for ADC_128SAMP:
```
2023-05-07 10:54:40,124 - DEBUG - INA219 write register 0x00: 0x07ff 0b0000011111111111
```
So still puzzled...

Chris

csreades, 2023-05-06:
My internet is down so I can look at this some more tomorrow. However what I was able to work out, and likely you already know, is that the device will let you read faster than the conversion rate it will just repeat old values. Is your read code looking for fresh data in someway?

chrisb2, 2023-05-06:
Figured it out, the CNVR bit must be checked using the `is_conversion_ready()` function, otherwise the bus register is read before all the conversions are done. I updated `perfomance-test.py` to:
```
#!/usr/bin/env python

import time
import logging
from ina219 import INA219

SHUNT_OHMS = 0.1
MAX_EXPECTED_AMPS = 0.2

READS = 100

ina = INA219(SHUNT_OHMS, MAX_EXPECTED_AMPS, log_level=logging.INFO, busnum=1)

def init():
    ina.configure(ina.RANGE_16V, ina.GAIN_AUTO, ina.ADC_9BIT, ina.ADC_9BIT)

def read():
    for x in range(0, READS):
        while (not ina.is_conversion_ready()):
            True
        ina.voltage()

if __name__ == "__main__":
    init()
    start = time.time()
    read()
    finish = time.time()
    elapsed = (finish - start) * 1000000
    print("Read time (average over %d reads): %d microseconds" %
          (READS, int(elapsed / READS)))
```
Then I get 126515us and 1721us for 128SAMP and 9BIT respectively.

Chris

csreades, 2023-05-06:
Brilliant, that makes sense then. Thanks for the info, I will incorporate that into my code, and thanks for the library!

closed 2023-05-06 by csreades